Sova-Tech/FTC-2020-2021 #16: Wheels

matei8, 2021-01-08:
Despre ce roti o sa folosim. Cand aveti o idee scrieti codul rotilor si de ce sa le folosim pe alea.

LucaVictor, 2021-01-12:
Folosirea unei suspensii deasupra intake-ului pentru downforce sporit

matei8, 2021-01-12:
Aprob ideea asta dar zic sa o punem in aplicare dupa ce facem teste si vedem daca e nevoie

matei8, 2021-01-15:
Avand in vedere ca acest issue a trecut de deadline
Pana maine vreau sa stabilim ce roti folosim la intake

matei8, 2021-01-15:
Eu zic sa folosim rotile de 30 mm  pt ca in afara de alea avem numai de 90 mm care sunt prea mari (30 mm -> REV-41-1353)

matei8, 2021-01-16:
Si plus ca avem in total 16 rotite
Si am sa fac acum niste calcule sa vedem cate ne-ar trebui in total

matei8, 2021-01-16:
Ok deci am facut calculele si o sa punem 8 rotite pe un ax custom probabil de 210 mm 
And that's that,  daca nu e bine, dupa ce facem teste o sa redechidem issue.

LucaVictor, 2021-01-19:
Avem o problema la rotile de 30mm. Modelul de CAD o sa va lamureasca
[image]
[image]
dupa cum se poate observa rotile mici nu au diametrul suficient de mare, iar tot sistemul de intake este expus. 
Avem 2 variante: ori folosim rotile alea de 60(le printam, le imprumutam nush cum facem) ori regandim tot sistemul de intake.

matei8, 2021-01-19:
at e o problema destul de mare

matei8, 2021-01-19:
din perspectiva mea putem face 2 chestii:
-Sa punem roti mai mari (60 mm) dar nu de 90 mm pt ca sunt prea mari
-Sau sa modificam structura

matei8, 2021-01-19:
NOTE: NU AVEM ROTI DE 60mm !!!

LucaVictor, 2021-01-19:
Chiar nu avem nicio roata de 60 ?
Nici macar una

matei8, 2021-01-19:
Nu... dar gasim altceva, cred cum stiu cum sa facem, o sa ma uit pe piese si am scriu

matei8, 2021-01-19:
Deci o varianta ar fi sa printam niste roti de 60 mm daca avem voie si dupa sa le dam cu ceva sticky pe suprafata 
#17

matei8, 2021-01-19:
> Deci o varianta ar fi sa printam niste roti de 60 mm daca avem voie si dupa sa le dam cu ceva sticky pe suprafata
> #17

@TakingTheRisk

TakingTheRisk, 2021-01-20:
Da frate se face

matei8, 2021-01-20:
Beautiful

Andrei-Iatagan, 2021-01-20:
În cazul în care nu găsim o modalitate de a folosi roți deja făcute, propun două variante alternative : 

i) (am văzut la o altă echipă ceva asemănător, link : https://www.youtube.com/watch?v=j5b6PV39Toc) Să folosim un ax cu niște tentacule flexibile, făcute din tub de perfuzie /elastic de praștie/ furtun.

ii) (nu am văzut în altă parte, deci nu sunt sigur cât de bine ar merge, dar cred că e ok) Să folosim un tambur în loc de roți, făcut dintr - o țeavă de PVC, cu două capace(să intre stea mai bine :)), acoperit cu o suprafață antiaderentă (cauciuc, heatshrink tubing, ceva spray ...)

Andrei-Iatagan, 2021-01-20:
#IDEI ?#
dacă are cineva vreo idee pentru a putea fixa și centra un tambur din țeavă PP pe un ax hexagonal ... este binevenit. (cred că puteți să „testați” acasă cu un tub de rolă de șervețele ...). 
tl, dr : dacă ști cum să pui tamburul mai bine pe ax, scrie aici

matei8, 2021-01-20:
> #IDEI ?#
> 
> dacă are cineva vreo idee pentru a putea fixa și centra un tambur din țeavă PP pe un ax hexagonal ... este binevenit. (cred că puteți să „testați” acasă cu un tub de rolă de șervețele ...). 
> 
> tl, dr : dacă ști cum să pui tamburul mai bine pe ax, scrie aici

Adica cum sa bagi axul sau cum sa il faci sa stea neclintit?

closed 2021-01-21 by matei8