Sova-Tech/FTC-2020-2021 #17: Motion system

matei8, 2021-01-08:
Cum o sa fie pus in miscare si ce componente vor fi folosite si de ce (se va da coudl produsului si motivul alegerii lui).

matei8, 2021-01-12:
[image]

matei8, 2021-01-12:
Asta ar fi in mare Intake systemul

matei8, 2021-01-12:
As vrea sa aduc tot bratul ala mai in fata pt ca dupa nu cred ca o sa mai incapa #18 (Conveyerul)

LucaVictor, 2021-01-13:
Incape la fix, si nu stiu ce modificari laterale mai putem aduce

costaalacuparmare, 2021-01-13:
Susțin acest design de intake si folosirea unor arcuri pentru suspensie

LucaVictor, 2021-01-13:
Folosirea unei suspensii deasupra intake-ului pentru downforce sporit ar fi o solutie, astfel incat sa folosim si rotile mai mici de 30mm

matei8, 2021-01-19:
> Folosirea unei suspensii deasupra intake-ului pentru downforce sporit ar fi o solutie, astfel incat sa folosim si rotile mai mici de 30mm

Cred ca mai intai trebuie sa facem teste sa vedem cum reactioneaza fara arcuri pt ca nu sunt chiar usor de  facut arcurile alea si ne ia din timp

matei8, 2021-01-19:
Bratul o sa il facem din extrusionuri de 225 mm (REV-41-1431) pt ca au cea ma decenta lungime pt ce ne trebuie noua

matei8, 2021-01-19:
Inca ma gandesc ce hex sa folosim si daca e nevoie sa taiem dintr-unul

LucaVictor, 2021-01-19:
Ar cam trebui sa taiem

LucaVictor, 2021-01-19:
Am creat urmatorul sistem de intake, insa are si el probleme lui, o sa postez aici un prototip sa va faceti o idee. 
[image]
Acesta este sistemul de tractiune, in schimb incercand sa rezolv problema anterioara, am dat in alta.
[image]
Se poate observa cum pur si simplu rotile nu reusesc sa atinga elementul de joc. O solutie ar fi sa ne printam rotile sau sa le folosim pe cele de 60

LucaVictor, 2021-01-19:
Voi ce credeti ?

LucaVictor, 2021-01-19:
Sistemul vechi merge, 
[image]
In schimb uitati cat de aproape este roata dintata de pamant/ elementul de joc
[image]
Voi ce parere aveti ?

matei8, 2021-01-19:
> Am creat urmatorul sistem de intake, insa are si el probleme lui, o sa postez aici un prototip sa va faceti o idee.
>
> [image]
>
> Acesta este sistemul de tractiune, in schimb incercand sa rezolv [...]

Esteopusul ideii de brat, pentru ca bratul ala e ca sa nu se miste tot robotul la interactiunea cu cercul, ci numai bratul

matei8, 2021-01-19:
Ramanem tot pe varianta cu bratul

matei8, 2021-01-19:
Trebuie sa schimbam putin structura bratului si o sa va zic imediat cum

matei8, 2021-01-19:
Deci rev-41-1303
[image]
Este din plastic si nu cred ca e o idee buna sa mentina axul care va primi atatea socuri, si oricum eu propun sa schimbam piesa asta cu unu cu margini mai mici

LucaVictor, 2021-01-19:
Nu cred ca asta ar fi o problema, in schimb daca am printa structura rotilor de 60 si le-am infasura intr-o banda elastica. Am ajunge la acelasi rezultat. Sau putem cumpara filament siliconat si sa printam rotile la dif densitati

matei8, 2021-01-19:
> Nu cred ca asta ar fi o problema, in schimb daca am printa structura rotilor de 60 si le-am infasura intr-o banda elastica. Am ajunge la acelasi rezultat. Sau putem cumpara filament siliconat si sa printam rotile la dif densitati

sa stii ca am putea face asta
@TakingTheRisk  crei ca poti face asta ?

LucaVictor, 2021-01-20:
Stelute + teava pvc pt intake 
[image]
https://www.youtube.com/watch?v=-ujskulDkoo

matei8, 2021-01-20:
Eu propun sa folosim un tub de pvc cu gauri hexagonale in capete

matei8, 2021-01-20:
Si putem sa il facem stick cu banda adeziva dubla, si putem sa o uzam putin inainte ca sa nu fie asa lipicioasa

LucaVictor, 2021-01-21:
Intake tip tambur 
[image]
[image]
[image]
Distanta fata de sol este aprox 72mm (71.897 mai exact sau 7,2 cm), am senzatia ca o sa ne trebuiasca mai mult downforce. Voi ce credeti ? @NegruAdam @TakingTheRisk @matei8 @Andrei-Iatagan @MateiBusoiu @pe_oricine_altcineva_am_uitat_sa_includ

LucaVictor, 2021-01-21:
Intake ramificatii:
[] tambur
[]stelute

LucaVictor, 2021-01-21:
Intake ramificatii 
-[ ] tambur
-[ ] stelute

LucaVictor, 2021-01-21:
Intake obiective:
- [ ] tambur
- [ ] stelute
- [ ] excursie la dedeman

matei8, 2021-01-21:
> Intake tip tambur
> [image]
> [image]
> [image]
> Distanta fata de sol este aprox 72mm (71.897 mai exact sau 7,2 cm), am senzatia ca o sa ne trebuiasca mai mult downforce. Voi ce credeti ? @NegruAdam @TakingTheRisk @matei8 @Andrei-Iatagan @MateiBusoiu @pe_oricine_altcineva_am_uitat_sa_includ

E ok, se putea sa aibe un diamentru mai mare dar merge si asa dac tot am cumparat deja

matei8, 2021-01-23:
Avem un 20 tooth sprocket (REV-41-1340) care este in plus pe partea dreapta a drivetrainului (efectiv nu are un rol in miscare) si pentru ca am folosit deja toate 20 tth sprocketurile o sa il scoatem p-ala de-acolo si inn locul lui o sa punem un spacer de 15 mm (REV-41-1323).

matei8, 2021-01-23:
In rest nu am depasit limita de piese asa ca e ok si deocamdata am terminat cu intake systemul si ne apucam de Conveyer #18

matei8, 2021-01-23:
Daca mai are cineva ceva de adaugat, in seara asta as vrea sa scrie aici.

closed 2021-01-30 by matei8