Sova-Tech/FTC-2020-2021 #18: Channel

matei8, 2021-01-09:
Despre portiunea dintre intake si Shooting System.

LucaVictor, 2021-01-12:
Folosirea unor roti care sa impinga inelele in shooter + un stabilizator 
[image]
Avand in vedere ca asta este drivetrain-ul nostru
[image]
Plus un stabilizator care sa centreze inelul pt shooter

matei8, 2021-01-12:
Astea sunt pt intake
Muta-le pe intake

MihaiBobeica, 2021-01-23:
Propun  folosirea rotilor de 30mm pentru conveyer, pentru stabilizator folosim elementul de intake.

matei8, 2021-01-23:
Prin stabilizator la ce va referiti ?

MihaiBobeica, 2021-01-24:
inelele când intră în teren sunt așezate vertical. Mă gândesc că un stabilizator are rolul de a le așeza pe teren, fiind un element incorporat in intake.

matei8, 2021-01-24:
Aici o sa discutam rezultatele finale de la #26 si #25

matei8, 2021-01-27:
Deci o sa folosim o rampa de policarbonat sau orice material (nu trebuie sa fie f rigid) ca sa indreptam inelele spre conveyer si ulterior shooting

matei8, 2021-02-06:
Avand in vedere ca nu e benefic sa folosim tuburile chirurgicale pt conveyer o sa folosim roti pe lateral care duc spre shooter

Intre ele va fi o panta de policarbonat
Tot sistemul va fi sustinut pe 2 C cahnneluri de 248 mm la un unghi de 45
Pareri ?

LucaVictor, 2021-02-06:
Piesa facuta in 3D care ne scuteste ne o gramada de piese si care in principiu coboara al doilea shaft pt a putea aduna piesele.
[image]
[image]
[image]
[image]

Credite @TakingTheRisk pt ca m-a ajutat sa o fac. Va rog frumos sa spuneti ceva ca sa vedem daca o printam sau mai facem modificari

matei8, 2021-02-06:
Eu cred ca e f buna piesa atata timp cat nu se mai freaca de ea shaft collar-ul

matei8, 2021-02-06:
dar cum vine lantul pe hex shaft ?

LucaVictor, 2021-02-06:
> dar cum vine lantul pe hex shaft ?

pun eu acm un sprocket

matei8, 2021-02-06:
> > dar cum vine lantul pe hex shaft ?
> 
> pun eu acm un sprocket

Unde ? ca nu prea are loc (btw trebuie sa fie aliniat si cu sprocketu de sus

LucaVictor, 2021-02-06:
> > > dar cum vine lantul pe hex shaft ?
> > 
> > 
> > pun eu acm un sprocket
> 
> Unde ? ca nu prea are loc (btw trebuie sa fie aliniat si cu sprocketu de sus

pot sa modific spatiul dintre rotile de 30

TakingTheRisk, 2021-02-06:
Eu tot n-am inteles dc a picat faza cu cablurile surgicale?

matei8, 2021-02-06:
Pt ca practic n-avem cum sa le prindem
Trebuia sa facem ca un fel de camaruta pt fiecare tub (ca sa aibe aderenta si sa nu se invarta in gol shaftu) pe fiecare ax si alea le faceam din Short Bearing-uri de care am putea sa mai printam dar o sa vb cu totii luni seara ka sedinta si luam o decizie impreuna

TakingTheRisk, 2021-02-06:
Negativ, nu am putea printa Short bearing-uri, nu pot printa cercuri.

Si nu ar trb neapara o camaruta tho? doar un canal si sa fie destul de intinse cablurile incat sa aiba frecare destula

matei8, 2021-02-06:
> Negativ, nu am putea printa Short bearing-uri, nu pot printa cercuri.
> 
> 
> 
> Si nu ar trb neapara o camaruta tho? doar un canal si sa fie destul de intinse cablurile incat sa aiba frecare destula
> 
> 

Problema e ca aluneca si n-au grip si daca nu poti printa cercuri, at facem cealalta varianta

matei8, 2021-02-12:
Am o idee pt cum sa construim conveyerul:

Dupa cum ati vazut, pe drivetrain sunt puse vertical doua extrusionuri (in dreapta si in stanga) si pe ele putem sa punem 15MM PLASTIC 120 DEGREE BRACKET ( REV-41-1311) si pt fiecare parte sa facem ca un brat de la intake 
[image]

closed 2021-02-23 by matei8