Sova-Tech/FTC-2020-2021 #19: Cum tragem ???

matei8, 2021-01-09:
Cum tragem  ?? :))

matei8, 2021-02-25:
https://www.joom.com/en/products/5cc6ee071436d4010143a679
Daca am pune motorul pe drivetrain(sub shooter anyway) si dupa am pune axul de la flywheel in asta am salva spatiu (sa nu iesim din parametrii)

TakingTheRisk, 2021-02-25:
[image]

O idee. Ce-ar fii sa folosim 2 roti(inegale ca altfel nu incape) ca sa reducem putin din bolta pe stanga? 
In chestia de mai sus e calculat sa fie si destula presiune pe inel ca sa fie destula frecare.

matei8, 2021-02-25:
Si adica sa fie conectat motorul si la ala ?

TakingTheRisk, 2021-02-25:
Ye, cu un lant pe deasupra.
Aprox. 490mm give or take 20mm

LucaVictor, 2021-02-25:
nu cred ca suruburile alea lungi o sa tina toata structura si motorul

matei8, 2021-02-25:
> Ye, cu un lant pe deasupra.
> Aprox. 490mm give or take 20mm

si care era limita ?

matei8, 2021-02-25:
> [image]
> 
> O idee. Ce-ar fii sa folosim 2 roti(inegale ca altfel nu incape) ca sa reducem putin din bolta pe stanga?
> In chestia de mai sus e calculat sa fie si destula presiune pe inel ca sa fie destula frecare.

stai ca acum am vazut defapt ce ai facut acum, and it's good, dar sa gasim cum sa facem sa aducem aia de plastic mai aproape de extrusion

matei8, 2021-02-25:
Am o idee care cred ca e destul de buna dar trebuie sa ma ajute cineva s-o bag in fusion

LucaVictor, 2021-02-27:
Acesta este abordarea mea
[image]

Sistemul se bazeaza pe 2 C channel-uri puse vertical legate cu 2 tipuri de 90 degree bracket
[image]

Sistemul shooter basic (atat pe robot cat si individual )
[image]
[image]

Am incercat sa gasesc si gauri pt suruburi, are 3 puncte dar nu formeaza un triunghi, asta mai ajustam manual
[image]

Cu putin ajutor se aliniaza perfect cu roata de 90

[image]

Mai multe poze: 

[image]
[image]
[image]

**Un dezavantaj** ar fi ca nu mai ramanem cu channel-uri dupa implementarea acestui sistem.

LucaVictor, 2021-02-27:
Astept opinii !!!!!!

matei8, 2021-02-27:
pare ok, dar lantul o sa vina deasupra pe la policarbonat nu ?
btw pune roti de 60 mm

LucaVictor, 2021-03-02:
La varianta pe care am zis-o eu cu shaftul de conveyer pe channel, am incercat ceva in fusion  si cam asta a iesit. Tin sa mentionez  ca diametrul tubului chirurgical calculat de mine in fusion ochiometric are 5.50mm diametru, iar distanta dintre pulley si channel este de 5.80mm 

[image]

Se aliniaza ok, cu extrusion-urile
[image]

Dar provocarea adevarata va fi sa alinem endcapul cu channel ul acolo
[image]

Astept  opinii !!!!!!!

LucaVictor, 2021-03-02:
As vrea sa stiu ce credeti ca sa stiu ce fac mai incolo, un flat plate ar fi perfect, poate intrebam pe xeo pe server poate ne ajuta cineva :))))))))))))

matei8, 2021-03-02:
Pune si tuburile ca am impresia ca intra in C Channel

Andrei-Iatagan, 2021-03-02:
pare ok, dar tot va fi nevoie sa punem ceva care sa faca inelul sa stea la acelasi nivel mereu.

matei8, 2021-03-02:
> pare ok, dar tot va fi nevoie sa punem ceva care sa faca inelul sa stea la acelasi nivel mereu.
> 
> 

Trebuie sa facem si o inclinatie cand ajunge la shooter pt ca nu trage destul de sus, ma gandesc sa facem o rampa de 3-4 cm (sau mai putin) dupa shooter

LucaVictor, 2021-03-03:
> Pune si tuburile ca am impresia ca intra in C Channel

tuburile in 3D sunt facute cam de forma, masuratorile nu sunt exacte, vb cu teo sa aflu exact masuratorile
In schimb dupa cum am zis, un flat plate ar fi perfect, asa ca as intreba pe xeo.
Din research-ul meu reiese ca nu ar trb sa ne fie frica sa folosim si piese custom din lemn sau plastic

LucaVictor, 2021-03-03:
> > pare ok, dar tot va fi nevoie sa punem ceva care sa faca inelul sa stea la acelasi nivel mereu.
> 
> Trebuie sa facem si o inclinatie cand ajunge la shooter pt ca nu trage destul de sus, ma gandesc sa facem o rampa de 3-4 cm (sau mai putin) dupa shooter

Asta nu e problema

LucaVictor, 2021-03-03:
Teo mi-a zis ca are aprox 3mm diametru, ceea ce este super si se portriveste

LucaVictor, 2021-03-03:
Dezavantejele in schimb ale acestui sistem sunt ca o sa avem nevoie de un shaft mai lung, se obs si in poza
[image]

Dar si tuburi chirurgicale mai lungi

LucaVictor, 2021-03-03:
Am pus ala si nu merge, teoretic am putea trage de tub, dar arata cam asa
[image]
[image]

Noi acum putem sa impingem de tub in jos pana se loveste de channel, dar nush  cat de eficient este :/

LucaVictor, 2021-03-03:
Am incercat sa ridic tot ansamblul o gaura mai sus, si chiar daca distanta dintre inel si tub s-a miscsorat,  inelul are acm mare damb de intampinat

[image]
[image]

matei8, 2021-03-03:
Eu zic sa lasam asta pt ca ne complicam si se modifica multe si plus ca merge si fara conveyerul extins diferenta e ca daca facem conveyerul pana in shooter nu ne mai trebuie rampa aia de policarbonat sau din ce o facem.

LucaVictor, 2021-03-03:
atunci vad doar o singura solutie, si aceea sa creeam ca niste canale, de preferat lemn care sa ghideze inelul spre roata
[image]

ceva in genul asta

[image]

putem sa facem cad-ul  astfel incat lemnul sa intre pe aici si sa aiba o pozitie oblica(efectiv ii dam slide pe extrusion)

[image]
 
iar mai sus la channel punem cum zice matei policarbonatul  ala ca sa nivelam.

[image]

matei8, 2021-03-03:
Yes, cred ca e cea mai eficienta chetstie pe care o putem face.

TeodorRuse, 2021-03-03:
Ca sa ridicam inelul, in loc de chestia cu carton si scotch care e acum, nu putem sa punem bucata aia de teava pe bara de jos, asa cum era inainte?

LucaVictor, 2021-03-03:
o incerc in fusion si iti zic, dar cred ca  da

LucaVictor, 2021-03-03:
> Yes, cred ca e cea mai eficienta chetstie pe care o putem face.

@TakingTheRisk te ocupi tu de niste schite ?

LucaVictor, 2021-03-03:
De asemenea, daca stie cineva niste locuri unde au CNC sau orice alta masina pt efectuarea unuei asemenea piese, este binevenit sa scrie aici

Eu am gasit atelierul asta dupa un search, dar cred ca avem optiuni mai bune: https://cnc-3d.ro/index.html

matei8, 2021-03-03:
> De asemenea, daca stie cineva niste locuri unde au CNC sau orice alta masina pt efectuarea unuei asemenea piese, este binevenit sa scrie aici
> 
> 
> 
> Eu am gasit atelierul asta dupa un search, dar cred ca avem optiuni mai bune: https://cnc-3d.ro/index.html

Stiu eu cui

LucaVictor, 2021-03-03:
> > De asemenea, daca stie cineva niste locuri unde au CNC sau orice alta masina pt efectuarea unuei asemenea piese, este binevenit sa scrie aici
> > Eu am gasit atelierul asta dupa un search, dar cred ca avem optiuni mai bune: https://cnc-3d.ro/index.html
> 
> Stiu eu cui, sa-mi dati schita cu piesa dupa ce e gata

Cat ne-ar costa si cat ar dura ?

matei8, 2021-03-04:
Nu stiu sigur sa va spun
Imi trebuie schita si dupa va spun, si cel mai probabil pt noi e moka :))

LucaVictor, 2021-03-04:
Am facut niste schite, o sa mai lucrez la curba aia, o sa mai modific niste detalii, dar asa arata in mare
[image]
[image]

Design usor de pus, dar trb sa scoatem shooterul deci nu e asa usor de pus :)))))))))
[image]

Cam asa  arata inelul in pozitia de tragere, e perfect pt ca il si strange putin, cel putin in teorie
[image]

Am largit putin bucata din dreapta ca sa-l si ghideze

[image]

LucaVictor, 2021-03-04:
Astept opinii

matei8, 2021-03-04:
Da e ok dar aia din dreapta nu trebuie neaparat dar stai asa ca acum mi-am dat seama... de ce nu printam piesa ca e mult mai rapid

matei8, 2021-03-04:
Defapt ba da e buna si aia din dreapta dar conteaza numai partea aia de langa flywheel

LucaVictor, 2021-03-04:
> Da e ok dar aia din dreapta nu trebuie neaparat dar stai asa ca acum mi-am dat seama... de ce nu printam piesa ca e mult mai rapid 

Pt ca are 22,5 cm, și efectiv consumam prea mult plastic degeaba

LucaVictor, 2021-03-04:
Nu cred că Alex are atâta spatiu

matei8, 2021-03-04:
> Am facut niste schite, o sa mai lucrez la curba aia, o sa mai modific niste detalii, dar asa arata in mare
> 
> [image]
> 
> [image]
> 
> 
> 
> Design usor de pus, dar trb sa scoatem shooterul deci nu e asa usor de pus :)))))))))

De ce trebuie scos shooterul ? Ca ala il montam pe conveyer nu ?
> 
> [image]
> 
> 
> 
> Cam asa  arata inelul in pozitia de tragere, e perfect pt ca il si strange putin, cel putin in teorie
> 
> [image]
> 
> 
> 
> Am largit putin bucata din dreapta ca sa-l si ghideze
> 
> 
> 
> [image]
> 
> 
> 
>

matei8, 2021-03-04:
Incerc sa printez eu pana luni

matei8, 2021-03-04:
@LucaVictor esti sigur ca inelul nu intra sub piesa, adica nu e prea mare spatiul dintre tuburi si piesa ?

TakingTheRisk, 2021-03-04:
> Nu cred că Alex are atâta spatiu

220x220x250, am destul

LucaVictor, 2021-03-04:
> @LucaVictor esti sigur ca inelul nu intra sub piesa, adica nu e prea mare spatiul dintre tuburi si piesa ?

Nu, piesa pleaca de la nivelul extrusion-uluiz deci nu ar trb să fie o problema, o sa dau o poza mai încolo sa vezi

matei8, 2021-03-04:
E ok ca m-am uitat si in fusion dar nu eram sigur daca erau masuratorile bune si d-aia te-am intrebat

closed 2021-03-13 by matei8