Sova-Tech/FTC-2020-2021 #21: Wabble Goal

matei8, 2021-01-09:


matei8, 2021-02-25:
are cineva idei pt wabble ?

LucaVictor, 2021-03-02:
O sa las aici poza discutata la sedinta de ieri

LucaVictor, 2021-03-02:
[image]

matei8, 2021-03-02:
1 si 6 ar fi cele mai bune puncte

MihaiBobeica, 2021-03-02:
Patratelele vor fi pe partea dreapta a terenului, deci am alege P6 pentru mecanism.

MihaiBobeica, 2021-03-02:
[image]
 ma gandesc ca ceva de genul sa facem, in punctul negru sa fie un ax, iar pe semicercul rosu avem un gear ce se misca, astfel pune popica in pozitie verticala, o ridica, iar apoi o coboara in patrat

matei8, 2021-03-02:
> Patratelele vor fi pe partea dreapta a terenului, deci am alege P6 pentru mecanism.

perfect

matei8, 2021-03-02:
> [image]
> ma gandesc ca ceva de genul sa facem, in punctul negru sa fie un ax, iar pe semicercul rosu avem un gear ce se misca, astfel pune popica in pozitie verticala, o ridica, iar apoi o coboara in patrat

True si atunci mai folosim un motor pt tot bratul ca sa-l ridicam ?

MihaiBobeica, 2021-03-03:
> > [image]
> > ma gandesc ca ceva de genul sa facem, in punctul negru sa fie un ax, iar pe semicercul rosu avem un gear ce se misca, astfel pune popica in pozitie verticala, o ridica, iar apoi o coboara in patrat
> 
> True si atunci mai folosim un motor pt tot bratul ca sa-l ridicam ?
> 

mă gândesc să folosim un motor ca să întoarcă acea roata π radiane. Nu înțeleg ideea de braț, cred că mai bine folosim o simplă roată ce se rotește atașată de robot daca înălțimea in '6' este cam cat înălțimea unui wobble

matei8, 2021-03-03:
Pai da ala e bratul, si in capat are o "gheara" care apuca wabbleul  si cand roata aia de care zici tu se roteste, bratul se ridica si wabbleul cu el

LucaVictor, 2021-03-03:
Putem  sa punem un core hex care roteste bratul pe axele Z,X(sus-jos), dupa care sa avem un servo  care activeaza ca o gheara  dupa cum a zis rancezu

LucaVictor, 2021-03-03:
Cred ca pe 6 mergem cel mai bine

MihaiBobeica, 2021-03-03:
Daca există o gheară nu avem nevoie de un motor in plus?

LucaVictor, 2021-03-03:
ba da, asta ziceam dar iti dau o poza
[image]

In  chenarul verde se observa, core hex ul si servo ul.
link video: https://www.youtube.com/watch?v=-ujskulDkoo 0:29

LucaVictor, 2021-03-03:
Core hex ul misca bratul sus jos, iar servo ul directioneaza gheara ca apuca

amareto1234, 2021-03-14:
multa bafta baieti !

closed 2021-03-16 by matei8