Sova-Tech/FTC-2020-2021 #24: Amortizor Intake

matei8, 2021-01-15:


Andrei-Iatagan, 2021-01-16:
propun arc lamelar, sustinut de o bara perpendiuclara pe podea, care sa fie pus pe bratul care tine rotitele de intake

Andrei-Iatagan, 2021-01-20:
în afară de arc lamelar, care am constatat că este o soluție complicată, dacă și numai dacă este nevoie de apăsare în plus pe inel la intake, putem să folosim elastice (chiar și elastice de birou), astfel : din șasiu scoatem un șurub, scoatem un șurub și din brațul de la intake, punem piulițe în capătul șuruburilor și efectiv întindem câteva elastice care să apropie cele două șuruburi
poză reprezentativă atașată
[image]

closed 2021-02-19 by matei8