Sova-Tech/FTC-2020-2021 #25: Intake --> Conveyer

matei8, 2021-01-24:
Trecerea din intake in conveyer(rampa+unghi+alte chestii ce mai e necesar)

Andrei-Iatagan, 2021-01-25:
https://www.youtube.com/watch?v=QHFo29VeIDY
(e prima echipa cu tambur pe care am vazut -o pana acum), si au un intake-: conveyer foarte simplu, cred ca e de urmarit

Andrei-Iatagan, 2021-01-25:
IMO, ăsta e cel mai bun sistem de departe (cred că am mai zis de vreo 5 ori, dar mai zic)
https://www.youtube.com/watch?v=F8CD3CI4OKc&t=99s
totul este foarte simplu, tragerea se face foarte simplu, nu e nevoie de eu știu ce braț care să împingă în flywheel, sau de o cutie de carton care să țină inelele. De asemenea, trecerea de la intake la shooting e foarte simplă, nu strică inele (adică nu există șansa să - l prindă la colț să - l îndoaie)

Andrei-Iatagan, 2021-01-25:
tl, dr : așa
[image]

Andrei-Iatagan, 2021-01-25:
încă o chestie pe care am observat - o : foarte muște echipe încep cu brațul de intake ridicat. S - a uitat cineva de la noi până acum pe cât de mare trebuie să fie robotul ? (parcă în anii trecuți erau niște limite, și lumea începea cu roboții pliați)

AndreiIatagan, 2021-01-25:
Sunt tot eu

costaalacuparmare, 2021-01-25:
Rampa va fi formata din banda cauciucata (vezi prastii piata) care sa traga inele, tuburi chirurgicale sau policarbonat

AndreiIatagan, 2021-01-25:
Având în vedere cele discutate astăzi în ședința referitoare la

AndreiIatagan, 2021-01-25:
În cazul în care vom avea nevoie de un „foraș” efectiv, am decis să folosim o placă din material plastic rigid, de exemplu policarbonat

closed 2021-01-25 by AndreiIatagan