Sova-Tech/FTC-2020-2021 #32: Stivuitor

matei8, 2021-03-11:
O sa fie deschis pana sambata la ora 12:00 (am). Ridicarea wabbelui de sub el.

MihaiBobeica, 2021-03-12:
In legatura cu Stivuitorul, voi pune mai multe imagini sa descriu ce am in minte

MihaiBobeica, 2021-03-12:
[image]
Dreptunghiul negru este robotul si partea albastra este elementul de wobble, probabil pus in spatele robotului

MihaiBobeica, 2021-03-12:
[image]
 Ma gandesc la un fel de lift, care sa deplaseze zona Galbena in Zona verde, probabil prin niste curele/gears. Ma voi gandi in detaliu despre cum se ridica

MihaiBobeica, 2021-03-12:
[image]
Ma gandesc sa atasasm o suprafata pe zona verde din ceva asemanator gear-ului, ca un o roata gear sa se roteasca sa duca in sus, Daca nu e suficient o roata, putem adauga mai multe, nu ar fi asta o problema. Ma gandesc la acest design de ceva ce iese in exterior pentru a evita depasirea inaltimii maxime

De asemenea ma gandesc ca zona pe care stationeaza Wobble se poate misca analog zonei verzi

MihaiBobeica, 2021-03-12:
[image]
 Stiu ca va plac desenele ( si ca sunt dificile de inteles), dar ma gandesc ca pentru a duce peste perete, stivuitorul sa fie plasat mai sus . Ma gandesc ca avanatajul la stivuitor este ca poate fi compactizat, urmand in timpul meciului sa se extinda la nevoie. Avantajul 2 ma gandesc ca este rapid si precis, bazandu-se in mare parte pe eficienta DriveTrainului

matei8, 2021-03-12:
Inca o chestie, ca sa mearga ideea cu stivuitorul, wabbleul trebuie sa ajunga deasupra robotului(ca sa fie mai inalt decat gardul adk 34 cm sau cv de genu)
Asadar, ma gandesc ca ne trebuie un sistem care sa functioneze intr-o raza de 35 cm

matei8, 2021-03-12:
> [image]
> Stiu ca va plac desenele ( si ca sunt dificile de inteles), dar ma gandesc ca pentru a duce peste perete, stivuitorul sa fie plasat mai sus . Ma gandesc ca avanatajul la stivuitor este ca [...]

nu cred ca poate sa fi foarte rapid pt ca trebuie sa tina wabbleul stabil

MihaiBobeica, 2021-03-12:
Un stivuitor industrial are barile puse paralel, mă gândesc că putem găsi un unghi astfel încât să stea mai fix acolo

closed 2021-03-13 by matei8