Sova-Tech/FTC-2020-2021 #35: Roti intake

matei8, 2021-03-23:
Spatiu dintre rotitele de la intake e ineficient si inelul se blocheaza acolo

Andrei-Iatagan, 2021-03-23:
Dupa parerea mea, daca am putea sa facem rost de tub chirurgical, aceasta problema ar putea fi rezolvata foarte simplu : practic, rotile de la intake vor deveni un conveyer mai mic. le inlocuim cu niste scripeti (care sunt destul de sigur ca pot fi printati 3D, pentru ca nu e nevoie de tensiune prea mare in tub).
    O alta varianta cred ca ar putea fi tentacule (sau ca am vazut de curand la neurobotix, pamatuf),care sa inlocuiasca efectiv rotitele. Practic ar avea efectul unor roti mai mari, dar nu sunt afectate asa de rau de distanta mica pana la pamant. dar nu sunt foarte sigur daca ar merge (trebuie sa ma gandesc putin mai bine)
    Inca o varianta posibila, care nu are nevie de piese suplimentare (poate de cateva rotite ca alea deja puse acolo), ar fi sa formam un fel de scripeti (ca la primul alineat), dar in loc de tuburi chirurgicale sa folosim elastice de birou (sunt constient de faptul ca le vom schimba destul de des, dar cred ca pentru 3 - 4 meciuri tin, si cand se rupe unul nu e mare paguba, daca sunt destule). [practic, unim rotitele 2 cate 2, nu e neaparat sa le lipim, poate fi facut si cu spacere, si intre cele 2 roti dintr - o pereche, sta cate un elastic, care, pentru o mai usoara instalare, va fi rupt, dupa legat (sa nu scoatem rotitele axle - urile cu roti). Care este avantajul peste chestia cu soricei de dinainte ? viteza elasticului va fi egala cu cea a rotii, si elasticul se va invarti, tragand si el de inel (nu va sta pe loc, doar sa nu cada inelul, cum era cazul la soricei)]

Andrei-Iatagan, 2021-03-23:
Si rog insistent pe Matei sa nu ma injure pentru issue - uri.

matei8, 2021-03-23:
Sunt de acord cu idee de mini conveyer plus ca asa o sa avem loc finally de 3 inele insa zic sa lasam rotitele alea de 30 mm pt ca propulseaza bine inelul

matei8, 2021-03-23:
Cred ca problema este defapt spatiul dintre cele doua axuri cu rotite unde propun sa punem niste curele mici care se prind pe niste rotite zimtate f mici pe care le avem in kit si sunt facute pt asta

Andrei-Iatagan, 2021-03-24:
<p>
legat de ce e mai sus : daca folosim rotite foarte mici si nu rotite cu acelasi diametru cu cele deja puse (sau poate le folosim direct pe cele de acolo), viteza miniconveyerului va fi mai mica decat viteza rotilor => inelul va fi incetinit. In cazul in care nimic nu merge, putem sa folosim roti mai mici.     
</p>

Andrei-Iatagan, 2021-03-24:
<p>
legat de curele, m - am uitat si eu pe optimus digital si pe ardushop. Toate curelele pe care le - am vazut eu pe acolo erau pentru imprimatre 3d, toate cam cu aceleasi dimensiuni, late de 10 / 6mm. Am mai gasit si asta, care pare folosibila, fiind atat de scurta (e inchisa) 
</p>

Andrei-Iatagan, 2021-03-24:
[image]

matei8, 2021-03-24:
> legat de curele, m - am uitat si eu pe optimus digital si pe ardushop. Toate curelele pe care le - am vazut eu pe acolo erau pentru imprimatre 3d, toate cam cu aceleasi dimensiuni, late de 10 / 6mm. Am mai gasit si asta, care pare folosibila, fiind atat de scurta (e inchisa)

da da da sunt bune pt ca am eu aici langa mine la imprimanta si au aderenta buna

matei8, 2021-03-24:
> legat de ce e mai sus : daca folosim rotite foarte mici si nu rotite cu acelasi diametru cu cele deja puse (sau poate le folosim direct pe cele de acolo), viteza miniconveyerului va fi mai mica [...]

Well u got a point dar, tot vor fi impinse spre conveyer si important e sa mai poata sa tina inca un inel

Andrei-Iatagan, 2021-03-24:
Nu este cea mai stralucita idee, pentru ca servo - urile sunt destul de sensibile, dar daca poate cineva imbunatati, astept raspunsuri
[image]

Andrei-Iatagan, 2021-03-24:
Acum intake - ul are doua probleme : sa nu ajunga prea jos (sa nu ajunga pe pamant), sa nu ajunga prea sus(adica sa nu poata intra mai mult de un inel odata pe sub el)

Andrei-Iatagan, 2021-03-24:
O posibila solutie pentru prima problema (sa nu ajunga intake - ul prea jos) este sa folosim doua sfori / suspante de lungime egala, care sa nu permita intake - ului sa ajunga pe sol. (atasez o poza, vor fi asemanatoare cu portile de castel ... :)  )
[image]

Andrei-Iatagan, 2021-03-24:
Intake - ul joaca foarte mult. Aceasta problema poate fi rezolvata fie prin limita superioara, fie prin cresterea masei intake - ului. O metoda pentru a pune greutate pe intake, ar fi sa printam 3D 2 piese in forma de U, care sa tina o teava / bara / Channel cu greutate variabila.
 Atasez o poza care sa descrie mai bine ce am zis

Andrei-Iatagan, 2021-03-24:
[image]

Andrei-Iatagan, 2021-03-24:
[image]

Andrei-Iatagan, 2021-03-30:
<p>Cred ca am mai gasit o mica optimizare pentru intake : punem soricei pe roti.</p>
<p>Atasez poza</p>
<p>Nu e ceva wow, dar cred că poate să facă intake - ul să meargă *puțin* mai bine</p>
[image]

Andrei-Iatagan, 2021-03-30:
[image]

Andrei-Iatagan, 2021-03-30:
valabil <=> un soricel are loc sa treaca de extrusion - ul pus intre cele 2 brate de la intake. Nu cred ca ar fi o mare imbunatatire.

closed 2021-04-06 by costaalacuparmare