Sova-Tech/FTC-2020-2021 #36: Tuburi si plexiglas

matei8, 2021-03-23:
Inelele se blocheaza in conveyer din cauza tuburilor si plexigkasului

Andrei-Iatagan, 2021-03-23:
Pentru a putea rezolva aceasta problema, cred ca avem 2 optiuni : 
i ) la tuburi : 
    a) crestem coef. de frecare al tuburilor (luam alte tuburi, de preferat din latex)
    b) atasam ceva pe tuburi pentru a opri alunecarea inelului (soricei pusi pe tuburi, elastice de bratari ...), avand totusi grija sa nu faca tuburile sa sara de pe rotite.
    c) apropiem tuburile de placa (merge numai daca este luat in calcul ce e scris la ii )
ii) la placa : folosim pentru placa un material foarte alunecos, precum policarbonatul (sau ce era pus inainte si ziceati voi ca mergea bine), sau chiar plexiglasul de acum, cu ceva lipit pe el (m - am mai gandit la cateva variante care pot fi direct atasate pe plexiglas : un mousepad:), tapet, folie de plastic (de la dosar), coperta de caiet, sticker translucid de se pune pe geamuri ...  ) 
ii )

Andrei-Iatagan, 2021-03-23:
Si rog insistent pe Matei sa nu ma injure pentru issue - uri

Andrei-Iatagan, 2021-03-23:
Am inteles si eu la ce ajuta invartirea conveyerului (robotul ar trebui sa traga mai precis acum), dar sa functionneze asa, cred ca e nevoie de ii)

matei8, 2021-03-23:
Eu zic sa folosim niste tuburi cu un coeficient de frecare mai ridicat si inca o chestie pe care am omis-o unii si care e destul de importanta e faptul ca atunci cand inelul sta PE tuburi si NU SUB forta de tractiune este mai mare pt ca frecarea dintre tuburi si inel e mai mare din cauza gravitatiei

matei8, 2021-03-23:
Si zic sa incercam treaba cu soricei

Andrei-Iatagan, 2021-03-24:
Oamenii au fost de acord sa folosim foaie de dosar :) yay !

closed 2021-04-26 by matei8