Sova-Tech/FTC-2020-2021 #41: optimizari

Andrei-Iatagan, 2021-04-24:


Andrei-Iatagan, 2021-04-24:
ce credeti ca poate fi imbunatatit ? mai important : cum

Andrei-Iatagan, 2021-04-26:
detectarea scenariului de joc cred poate fi îmbunătățită prin software : algoritmul poate să „caute” inele într - o zonă mai mare (spre exemplu prin verificarea mai multor pixeli pe fiecare rând, astfel încât să nu fie necesară alinierea perfectă)

Andrei-Iatagan, 2021-04-26:
cred că putem să facem îmbunătățim shooterul prin modificarea legăturii între el și conveyer. (facem astfel încât inelul să intre de fiecare dată la fel în shooter, deoarece în momentul actual mai apare câte un „misfire”, de obicei în autonomie)

Andrei-Iatagan, 2021-04-26:
cred că ar ajuta ca arbitrii (ei știu regulile foarte bine) să facă un fel de „instructaj” cu cei care participă la meciuri, poate ajută la evitarea unor viitoare penalizări.

Andrei-Iatagan, 2021-04-26:
pentru a face shooter - ul să nu mai piardă atât de multă putere după ce trage un inel, cred că putem să punem greutăți în flywheel (sau să schimbăm flywheel - ul curent cu ceva făcut dintr - o singură piesă, fapt ce ar ajuta si cu misfire - ul). (ca și greutăți : bucăți de ax tăiate, bucșe mici, puse în găurile de la exteriorul roților, disc metalic (găurit să intre axul ) lipit între ele, chiar și o greutate pusă peste cchanel, unde este camera acum
)

MihaiBobeica, 2021-04-26:
Propun ca pentru a scadea puterea shooter-ului, vom luat un disc metalic (mentionat de @AndreiIatagan ) si il vom pune intre cele doua fly-wheel-uri pentru a creste inertia sistemului. Cum inertia este mai mare, atunci vom putea sa nu avem un curent care curge continuu in shooter, (de exemplu sa sara peste M3)

matei8, 2021-04-26:
> detectarea scenariului de joc cred că poate fi îmbunătățită prin software : algoritmul poate să „caute” inele într - o zonă mai mare (spre exemplu prin verificarea mai multor pixeli pe fiecare rând, astfel încât să nu fie necesară alinierea perfectă)

Nu asta e problema, problema e ca nu e bine pozitionata camera

closed 2021-07-26 by matei8