Sova-Tech/FTC-2020-2021 #45: Probleme Intake

matei8, 2021-05-10:


MihaiBobeica, 2021-05-10:
Propun sa renuntam la intake separat de conveyer. Prelungirea conveyer-ului ar fi o idee buna... ( se realizeaza cu conveyer-ul peste inel, nu cum este acum sub)

Andrei-Iatagan, 2021-05-10:
<h4>inel sub intake</h4>
<p>uneori, daca se face wheelie inainte sa se ia un inel, acel inel intra sub conveyer si este greu de scos. putem sa punem o placa astfel incat sa nu se mai intample asta (2*L, dreptunghi lung, prins de sasiu)</p>
<h4>intake -> conveyer</h4>
<p>nu merge, e  nevoie de wheelie 1/2 inele, ceea ce nu e bine (tbh, nu stiu acum cum sa rezolv asta)</p>
<h4>randament mic</h4>
<p>sunt de acord cu Mihai. prelungirea conveyer - ului ar scoate multe lanturi, care consuma foarte multa energie</p>

matei8, 2021-05-10:
> Prelungirea conveyer-ului ar fi o idee buna... ( se realizeaza cu conveyer-ul peste inel, nu cum este acum sub)

Daca prelungim conveyerul il prelungim pana la al doilea ax

Andrei-Iatagan, 2021-05-10:
inlocuim partea proasta cu o rampa

costaalacuparmare, 2021-05-10:
sper sa avem destul tub incat sa facem conveyerul de la al doilea ax al intakeului la shooter. asta ar rezolva problema gaurii dar nu stiu cum am ridica atunci intakeul

matei8, 2021-05-10:
> sper sa avem destul tub incat sa facem conveyerul de la al doilea ax al intakeului la shooter. asta ar rezolva problema gaurii dar nu stiu cum am ridica atunci intakeul

il facem f scurt cum il aveau cei de la quantum

matei8, 2021-05-10:
> > sper sa avem destul tub incat sa facem conveyerul de la al doilea ax al intakeului la shooter. asta ar rezolva problema gaurii dar nu stiu cum am ridica atunci intakeul
> 
> ne-am gandit la asta. vom folosi o rampa de la sol la conveyer ( eventual punem niste roti peste pentru tractiune). o sa completeze Matei cu detalii

well rampa ar incepe la vreo 2 cm de rotile de intake ca atunci and inelul se afla pe rampa sa aibe cat mai multa viteza, datorita caruia va ajunge pana la conveyer. 

In cazul in care inelele totusi nu urca o sa facem ideea lui Bobeica de a mai pune un ax cu rotite deasupra rampei ca sa continuam sa miscam inelle spre conveyer de unde vor fi preluate de soricei care vor ramane pe tuburi pt chiar au ajutat f mult

matei8, 2021-05-14:
[image]

matei8, 2021-05-14:
[image]

matei8, 2021-05-14:
> [image]

Aici ar veni rampa

closed 2021-07-26 by matei8