Sova-Tech/FTC-2020-2021 #46: Probleme conveyer

matei8, 2021-05-10:
Este scurt si este prea departat de intake, unde cad inelele

matei8, 2021-05-10:
In cazul asta o sa renuntam la al doilea ax  de la intake (cel dinspre conveyer) si sa prelungim tuburile pana pe el, si probabil sa dam axul respectiv putin mai jos ca sa ni fie probleme dinspre intake spre conveyer la preluarea inelului

MihaiBobeica, 2021-05-11:
Cu aceste schimbări ar trebui și să schimbăm unghiul pe care îl face conveyer cu solul (să îl ducem de la 30° actual la 42° «unghiul râmpii»)

matei8, 2021-05-11:
> Cu aceste schimbări ar trebui și să schimbăm unghiul pe care îl face conveyer cu solul (să îl ducem de la 30° actual la 42° «unghiul râmpii»)

Daca facem asta o sa intre in C Channel-ul de mijloc de la Drivetrain

matei8, 2021-05-14:
O sa dam putin mai in spate al doilea ax ca sa nu fie diferenta prea mare de unghi cand trece de la intake la conveyer

matei8, 2021-05-14:
[image]

closed 2021-07-26 by matei8