Sova-Tech/FTC-2020-2021 #49: Battery Tray

NegruAdam, 2021-05-10:
Cum facem un sertar pentru baterie pe care sa il putem trage pentru a schimba bateria fara sa ne incurce plexiglas-ul?

matei8, 2021-05-10:
Putem sa facem niste piese care se monteaza pe cele doua extrusionuri, si intre ele o placa ca un raft pe care sa stea bateria si sa fie si sus si jos

MihaiBobeica, 2021-05-11:
Totuși ar trebui sa ridicam destul de mult în sus ca să putem folosi o șurubelniță (nu o cheie)

matei8, 2021-05-11:
Pai o data montat acolo ramane si nu o sa mai fie nevoie sa o desurubam

matei8, 2021-05-12:
Ideea acestui "sertar" este sa nu mai trebuiasca sa folosim suruburi, o sa se puna bateria in sertar, care se inchide cu o usita or anything si dupa se conecteaza la switch. Nu o sa fie nevoie de suruburi

closed 2021-07-26 by matei8